alice-fit-fee-upgrade/FIT_PM_Dual_Range_AFE #1: Option for extended time measurement

skoryciak, 2025-05-19:
If there is space available on the PCB, please extend the time measurement path for experiments with SERDES on FPGA. The SN65LVDS100 (DD17) chip can be replaced with the SN65LVDS250. All four outputs from this crosspoint switch must be connected to a single FPGA bank (64 seems to be occupied, co maybe 65?). The first input to the switch will be CFD_OUT, and the second one will be a LVDS signal from the FPGA from the same bank (for calibration). Control should be done from the FPGA level to allow sending synchronization sequences during operation.

filipswit, 2025-06-08:
I can try to fit it, but I don't promise I will.

Do I understand well your needs?
[image]

filipswit, 2025-06-28:
Added.

closed 2025-06-28 by filipswit